priseborough/InertialNav #16: NavFilterTestHarness24': Undefined function or variable 'startTime'

shening, 2015-07-02:
Hi 
I am getting the error below when trying to run NavFilterTestHarness24.mdl in Matlab 2015a. I have followed the steps in the readme. Any clue why its happening ?

Thank you

Error evaluating expression 'startTime' for 'StartTime' specified in the Configuration Parameters dialog for block diagram 'NavFilterTestHarness24': Undefined function or variable 'startTime'

adityadeshpande2010, 2016-06-15:
I am having the same problem as shening. Can anyone help.?
Thanks.

rafaelmendy, 2017-01-06:
after changing your current directory to models. run this script in matlab 
`addpath('../scripts/','../TestData')`

esridhar159, 2017-02-02:
I am also facing the same problem as( **Undefined function or variable 'startTime**'.) I am totally confused about this error. Please someone help me to solve this error.

rafaelmendy, 2017-02-02:
@esridhar159 
I think if you do the procedure that i mention in the above comment. this problem will be solved.
1) change your directory to models. ( as mentioned in the readme)
2) run`addpath('../scripts/','../TestData')`
3) run `RunNavFilterTestHarness24.m` (not RunNavFilterTestHarness24.mdl)

snn-Lenovo-G405s, 2017-03-19:
@rafaelmendy why did I follow you,there was error  ,like this
Error using load
Unable to read file 'simtest20.mat': no such file or directory.
I don't know where  is simtest20.mat.

rafaelmendy, 2017-03-19:
@snn-Lenovo-G405s 
Hi
You are right. there is no such file "simtest20.mat"
So, you must use other test data that exist in the "TestData" folder. 
open LoadNavFilterTestData.m  and comment "load('simtest20.mat')" and uncomment another one.
for example:
`load('TestLog1.mat')`
`%load('TestLog2.mat')`
`%load('TestLog3.mat')`
`%load('simtest20.mat')`

snn-Lenovo-G405s, 2017-03-19:
@rafaelmendy Thank you .The problem is solved