priseborough/InertialNav #20: Running the Simulink Model

SainiVinod, 2016-11-30:
Hi,

I am not able to run the simulink model. I tried executing RunNavFilterTestHarness21 file in rapid accelerator model, but it throws "Unable to build a standalone executable to simulate the model 'NavFilterTestHarness21' in rapid accelerator mode.". I had the same problem running with RunNavFilterTestHarness24. Could you help me out ?

Thanks. 

Regards,
Vinod

xiaoqiaomu, 2016-11-30:
I am sorry ,i don't know either

发自网易邮箱大师
On 11/30/2016 13:45, Vinod Saini wrote:

Hi,

I am not able to run the simulink model. I tried executing RunNavFilterTestHarness21 file in rapid accelerator model, but it throws "Unable to build a standalone executable to simulate the model 'NavFilterTestHarness21' in rapid accelerator mode.". I had the same problem running with RunNavFilterTestHarness24. Could you help me out ?

Thanks.

Regards,
Vinod

zhujimson, 2016-12-07:
Did you guys exactly do the right steps to build the M code? And the IDE should later than matlab2012.If you do, it should be fine.

zhujimson, 2016-12-07:
@SainiVinod

SainiVinod, 2016-12-07:
thanks @zhujimson. Its running properly when run in normal mode, but there seems to be some issue when it is being executed in accelerator mode. It might be because of compiler, I'll try to figure it out. 

Thanks !

Tearabbit317, 2016-12-09:
This is what I am getting for the error... anyone knows how to fix it?

Undefined function 'Quat2Tbn' for input arguments of type 'sym'.

Error in EKF (line 85)
Tbn = Quat2Tbn([q0,q1,q2,q3]);

rafaelmendy, 2017-01-06:
Hi
probably  you've not add the folder that contains 'Quat2Tbn' function in your matlab directory. add the folder and it will be solved.

closed 2017-04-13 by SainiVinod