priseborough/InertialNav #21: Why the magMeas is the sum up of inertial frame mag and body frame mag

xeonqq, 2016-12-14:
Hello, Paul
I am very inteseted in your work and had a little doubt here.

As shown in the matlab code, here you have
```
%% derive equations for fusion of magnetic field measurement
magMeas = transpose(Tbn)*[magN;magE;magD] + [magX;magY;magZ]; % predicted measurement
```
I think the magnetometer sensor only measures  [magX;magY;magZ], which is earth mag field in body frame.
so I think
```
magMeas = transpose(Tbn)*[magN;magE;magD]
```
for the observation function would be correct

and then 
```
innovation = magMeas - [magX;magY;magZ]
```
Could you explain a little about this?

rafaelmendy, 2017-01-06:
Hi
I think the magnetometer measures all magnetic fields around itself and not only magnetic field of earth.
So, this formulation has supposed that there may be a specific kind of magnetic field in body frame. for example magnetic field of brushless motors or servos. magnetometer measures them as well. 
So it's the correct and in my thought talent job to let the algorithm compensate and self calibrate these kind of magnetic fields.

xeonqq, 2017-01-08:
@rafaelmendy Thank you very much, I think what you said makes sense, it finally clear my doubt on this. But just another thing pop up on my mind, if EKF can estimate the mag noise/offset, why Ardupilot would still need manual mag calibration. Maybe to provide a better initial guess of the value?

rafaelmendy, 2017-01-08:
Yes
I think so.

closed 2017-01-10 by xeonqq