priseborough/InertialNav #41: Stuck in [PP,SPP]=OptimiseAlgebra(PP,'SPP');

huyang8911, 2023-02-09:
Matlab R2021a, run m files "GenerateNavFilterEquations.m", stuck in [PP,SPP]=OptimiseAlgebra(PP,'SPP'); I have spend about 30hours run this code. The index loop to 7800, and still not finish. I have updated the function "OptimiseAlgebra". Any one known where is wrong?
[image]

average-engineer, 2024-01-03:
Somehow in the newer versions, the OptimiseAlgebra script gets stuck in a loop while establishing the subexpressions for the covariance matrix (`[PP,SPP]=OptimiseAlgebra(PP,'SPP')`). What worked for me was changing the condition `if f_complete || index > 100` to `if f_complete || index > 500`. After this change, my script successfully runs in about 2.5 hours

huyang8911, 2024-11-21:
> Somehow in the newer versions, the OptimiseAlgebra script gets stuck in a loop while establishing the subexpressions for the covariance matrix (`[PP,SPP]=OptimiseAlgebra(PP,'SPP')`). What worked for [...]

Great job! I'm try and change to "if f_complete || index > 500" ,than it works.